OrcaSlicer/OrcaSlicer #691: no more thumbnails

AIIrounder, 2023-04-07:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

AIIrounder, 2023-04-07:
I see no thumbnail and info more with version 1.6.1
[image]

hliebscher, 2023-04-07:
@AIIrounder  bei welchem Drucker ? Klipper ? welches Betriebssystem?

AIIrounder, 2023-04-07:
Drucker ist ein Voron mit der Firmware Klipper und Betriebssystem wäre dann Win 10.

hliebscher, 2023-04-07:
sollte gefixt sein in der aktuellen 1.6.1  bei mir sind die zu sehen und im Gcode auch drin... gab einen fix 
fix Malformed command issue for Klipper am 6.4 9:23
oder der alte Mainsail bug...
und wichtig 
If you're running Klipper, it's recommended to add the following configuration to your `printer.cfg` file.
```
# Enable object exclusion
[exclude_object]

# Enable arcs support
[gcode_arcs]
resolution: 0.1
```

SoftFever, 2023-04-07:
Please update Moonraker to the latest version.
I put a note on 1.6.0 release page.
Let me put this message in 1.6.1 release page as well.
[image]

AIIrounder, 2023-04-07:
it was moonraker

closed 2023-04-07 by AIIrounder