OrcaSlicer/OrcaSlicer #692: Filament Settings not saving

Quick-Flash, 2023-04-07:
**Describe the bug**
Changes I have made to filament profiles do not always save. One particular PLA profile constantly gets reset and I have to manually save it again upon reopening OrcaSlicer, it doesn't seem to effect other filament profiles however. These settings also do not seem to transfer between versions of OrcaSlicer, which makes me not want to update as I have to re-input my filament settings.

**3mf File for This Bug**
Not relevant

**To Reproduce**
Have multiple OrcaSlicer filament profiles (ensure you have set your own pressure advance) and then close OrcaSlicer and reopen it to find that your saved profile is there, just without your changes.

**Expected behavior**
Able to save custom filament profiles.

**Screenshots**
Not needed.

**Printer model**
X1c

**Desktop (please complete the following information):**
Windows 10
Happened in OrcaSlicer 1.5 and 1.6

ajalberd, 2023-06-01:
This happens to me, as well. All my settings reset after every launch except printer setting.

julie777, 2023-06-01:
Do you have the preferences setting to sync setting turned on?

closed 2023-11-20 by github-actions[bot]