OrcaSlicer/OrcaSlicer #693: Starting temperature not set for temperature calibration

BrutalEngineering, 2023-04-07:
**Describe the bug**
When running the temperature calibration the print starts without warming the nozzle to the 1st layer temp. In our case the nozzle is @140°C instead of 320°C

**3mf File for This Bug**
Attached 3mf file

**To Reproduce**
Send to Bambu Labs x1c and print. doesn't matter if auto leveling is on or off.

**Expected behavior**
Material will be loaded an purge properly so nozzle is not clogged. Printer will run through bed leveling and start to print with nozzle at 140C when the first layer should be 320C

**Screenshots**
Printer scans first layer and fails because nothing is printed. no filament is extruded on the bed.

**Printer model**
Bambu Labs X1C with AMS
**Desktop (please complete the following information):**
 - OS: Windows 10 
 - Version 21H2, Orca 1.61
[temperature_tower_PA-CF_1h43m.zip](https://github.com/SoftFever/OrcaSlicer/files/11180170/temperature_tower_PA-CF_1h43m.zip)

jaikdean, 2023-06-09:
I'm having the same issue in OrcaSlicer 1.6.3 on macOS. My X1C's firmware is the current latest of 01.05.01.00. I've tried the filament in positions 1 and 3 in the AMS with no effect.

fsahmad, 2023-08-27:
Just ran into the same issue trying to calibrate PET-CF which defaults to 320C as maximum temperature

Zodiase, 2024-02-24:
Seeing the same problem. What's the workaround?