OrcaSlicer/OrcaSlicer #695: The settings to visually customize the bed don't work

liftbag, 2023-04-07:
**Describe the bug**
The settings to visually customize the bed don't work. It was opened as a feature request in the issue #685, but I think that since the option is not working, it should be reported as a bug. 

As a first step, it would be very useful to add a custom printer among the available options, as well as on PrusaSlicer, without necessarily having to start from a default profile (to be modified later).

[image]

**To Reproduce**
1. Go to Printer edit preset
2. Open the Prtintable area settings
3. Select a texture and model from anywhere on your computer.
4. Return to the "Prepare" workspace

**Expected behavior**
The bed should show the shape of the loaded STL and display the top surface texture.

[image]

To obtain the result of the screenshot above, every time Orca Slicer is updated I replace the two files listed below

[image]

Note that the texture is not shown even with this method. The bed should look like this (in PrusaSlicer the texture is white).

[image]

**Printer model**
Totally custom.

**Desktop (please complete the following information):**
 - OS: macos Ventura 13.3 (Intel)
 - Orca Slicer Version:
 
[image]

H4kunaM4tata, 2023-04-08:
@liftbag I thought it was something specific, that is why I tagged it as feature.
I've changed it to bug.

liftbag, 2023-04-08:
> @liftbag I thought it was something specific, that is why I tagged it as feature. I've changed it to bug.

Ok. I hope this problem will be resolved soon, because I use a totally custom printer, and it would be great not to have to partially reconfigure it with each software update.

liftbag, 2023-04-15:
It works in the latest dev build. Going to close.
Thanks a lot.

liftbag, 2023-04-15:
@SoftFever I reopened because OrcaSlicer does not store model and texture paths on program exit. Upon reopening, the customizations are lost. 
I opened the issue working on a mac intel, while currently I switched to Apple silicon. Maybe it could be a problem related to this platform?

liftbag, 2023-04-15:
> Maybe it could be a problem related to this platform?

Or is the problem related to the lack of a totally custom profile as a starting point?
I built my custom profile starting from a Prusa MK3. Although renamed, the profile somehow recalls the MK3, for example by showing the ORIGINAL PRUSA I3 silkscreen on the bed.
Until now, to obtain a permanent change to the profile, I physically replaced the mk3_bed svg and stl files in the application folder. With the latest build, when the program is reopened, the customized model recovers the original stl and svg files of the profile, which this time I have not replaced.

SoftFever, 2023-04-16:
@liftbag did you save the profile after assigning bed texture/model?

liftbag, 2023-04-16:
> @liftbag did you save the profile after assigning bed texture/model?

Yes, I make sure the asterisk disappears, then to be safe I save it again. I work on it and eventually close the program. When I reopen, it has the Prusa bed and the references to the custom model are lost.

SoftFever, 2023-04-18:
Fixed in 1.6.2 beta

closed 2023-04-18 by SoftFever