OrcaSlicer/OrcaSlicer #697: volumetric flowrate model has errors

julie777, 2023-04-07:
**Describe the bug**

The model that is generated for the max flowrate test has errors.

[image]

**To Reproduce**
Steps to reproduce the behavior
1. select calibration/more/max flowrate
2. model will load and error appear

**Expected behavior**
model should not have errors 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

steezyprintz, 2023-04-08:
I have had that issue but also that the flow change doesnt get applied and also the VFA speed version too, have to close and restart a few times to get it to work again

Navyguy330, 2023-05-28:
> I have had that issue but also that the flow change doesnt get applied and also the VFA speed version too, have to close and restart a few times to get it to work again

yeah, came here for that.  When I use the Max Volumetric speed test, it does not actually do anything. I have had it to 50mms/3 and it came out the same as at 5 lol.

closed 2023-11-20 by github-actions[bot]