OrcaSlicer/OrcaSlicer #711: plate labels not in preview

crushedpadding, 2023-04-09:
**Describe the bug**
1.5.0 showed plate type in preview, ex cool plate, high temp, etc.
1.6.1 does not and caused error when printer trying to level and prep to print. in screenshot see label on left missing along with bottom.

**3mf File for This Bug**
n/a

**To Reproduce**
open orca, change plate time, look at preview

**Expected behavior**
plate preview shows expected plate image like old version

**Screenshots**
new:

[image]

old:

[image]

**Printer model**
x1c

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] windows
 - Version [e.g. 22] 10

crushedpadding, 2023-04-09:
@SoftFever i think is bug not enhancement since this was old behavior in 1.5 and removal not in change notes.

SoftFever, 2023-04-09:
You can switch back to the older behavior by turning off "Show auto-calibration marks" in printer settings.
Which will display plate type

closed 2023-11-19 by SoftFever