OrcaSlicer/OrcaSlicer #712: Unable to preview GCODE files

H4kunaM4tata, 2023-04-09:
TLDR: If you need to load a GCODE file to check if you have the right profiles in place, that won't happen forcing you to slice everything again just to be sure.

OS is unable to load its own GCODE for preview:

[image]

But the file was sliced via OS and ironically, if I try to load it on PS, it loads, broken but loads.

[image]

This is a GCODE file sliced via PS.
This is what you expect to see with OS

[image]

Side note: OS does not have "Load GCODE" like PS but drag and release.

julie777, 2023-04-24:
Gcode files should have variables defined (or placeholders as BL calls them) for the type of printer and other information (plate type, etc.) to indicate whether they are usable or not for a given printer.

H4kunaM4tata, 2023-04-25:
> Gcode files should have variables defined (or placeholders as BL calls them) for the type of printer and other information (plate type, etc.) to indicate whether they are usable or not for a given printer.
> 

This is one of the most annoying problem, buy have to slice things over and over because it's unable to load a GCODE file like PrusaSlicer

R4toLoko, 2023-05-22:
@SoftFever the problem has not been solved on Linux at least and 1.6.3 beta.

[image]

SoftFever, 2023-05-22:
@R4toLoko 
Can you share your gcode?
Mine works fine.
[111.zip](https://github.com/SoftFever/OrcaSlicer/files/11537709/111.zip)

R4toLoko, 2023-08-08:
@SoftFever I am so sorry for the delay on this.
I tried to load your gcode file and nope.
Let me ask others.

[image]

R4toLoko, 2023-08-16:
This appears to be a Linux bug, Mac seems to be working just fine.
Confirmed with other users.

R4toLoko, 2023-11-08:
This has been completed, tested and it is working. 
Thanks a lot.

closed 2023-11-08 by R4toLoko