OrcaSlicer/OrcaSlicer #716: Set Brim color in multi color prints.

retoid, 2023-04-09:
**Is your feature request related to a problem? Please describe.**
Currently there doesn't seem to be a way to set the brim color in multi color prints. Or to determine which color is printed first when multiple colors are on the first layer. This is an issue when the brim color is touching parts of a print that are printed in a different color.

**Describe the solution you'd like**
A feature/selection to set brim color/material just as there is one to set support color/material.

**Describe alternatives you've considered**
n/a

**Additional context**
Thank you

Gridzilla, 2024-01-04:
Mouse ears in accent colour. This is completely random and is not affected by order of filament.
[image]

closed 2024-04-12 by github-actions[bot]

DonFelton, 2024-05-11:
I am also having issues with this brim coloring. Printing signs and the brim is coming out in the letter color not the backing plate color.