OrcaSlicer/OrcaSlicer #717: [FEATURE REQUEST] Measure tool

H4kunaM4tata, 2023-04-10:
Is there any chance of having this measure tool implemented on OS??

It allows you to select a vertex, an edge, radios or a whole face, making it easier to use the slicer itself to make precise boolean operations without having to use a 3D tool such as FreeCAD or Blender.

[image]
[image]
[image]
[image]

Thank you

chrbratt, 2023-04-30:
+1 for this.

Very useful when rescaling models and still want to achieve a certain fit for something. :)

cybrsage, 2023-07-05:
I would LOVE this - I dislike trying to guesstimate this stuff...I am usually slightly wrong.

DrKlipper, 2023-09-22:
This is also a perfect tool if you reverse engineer some STL files. 
Especially when you measure distances between two holes.

JurajMik, 2023-10-23:
I have Prusa Slicere just for that. Would be happy to have this in Orca.

R4toLoko, 2023-11-08:
This is currently being worked on, beta version is available and almost ready for the stable release.
Any issue, testing is being added to that PR.
Closing this ticket.

PR #2603 (2023-11-19): Port PrusaSlicer's Measure & Cut gizmos
Build on top of #2520 and replaces it.

This PR includes:
- Rewrite the opengl rendering code, which now renders (almost) everything using shaders instead of legacy opengl function calls
- Rewriting the gizmo mouse handling code that moves the mouse handling coding into each gizmo themselves
- Rewriting the mouse picking code, now it uses ray casting to figure out what's under the mouse cursor
- Porting of the PrusaSlicer's measure tool
- Replacing existing cuting tool with the better one PrusaSlicer has
- Updating of other gizmos using PrusaSlicer's lat [... cut]
closed 2023-11-08 by R4toLoko