OrcaSlicer/OrcaSlicer #719: [Frature Request] BedLevelVisualizer

Obicom, 2023-04-10:
In regards of bending heat bed untortunately the auto bed leveling values are not visible

If  possible it would be great to get values or better a bed visualisation to be ble to align the bed.

Take a look how it cloud look like:

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer

wociscz, 2023-04-12:
I'd say this is not possible, due to encrypted logs from the printer - basically we haven't such data from the printer (bed mesh) available - only Bambulab staff are available to "debug" it from it's logs.
Unless someone decrypt the log files (find the way how to decrypt them).

hliebscher, 2023-04-14:
i think this is an printer independent slicer and is in most printer interfaces

closed 2023-11-20 by github-actions[bot]