OrcaSlicer/OrcaSlicer #721: OrcaSlicer crashes very frequently on Windows 10

julie777, 2023-04-10:
**Describe the bug**

Orca Slicer has been open and running. I am working on something else and it is in the background. I try to switch Windows to load a new model, or check print progress, etc. and I find that it is no longer running.

On the positive, when I restart it I am asked if I want to recover the project that I was working on.

Diagnosis

- It doesn't crashing while I am actively using it to slice or anything else.
- I am not sure if the camera is on play all the time when it happens, but I will try to keep track of that from now on
- there is nothing in the debug log or the network log to indicate a reason. The log entries are just terminated in the middle of an entry and the entries are never the same, so there is no indication that it is doing the same thing each time
- 

**To Reproduce**

I do not have a way to quickly cause the crash.
My suggestion is to just go to devices, connect to a printer and start monitoring the camera and then do something else letting the slicer window go to the background.

**Expected behavior**
Applications should never crash.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
P1P

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.6.1

julie777, 2023-04-13:
I have a possible cause for the crash.

Bambu Slicer was hanging very frequently. It turned out to be a compatibility problem with the NVidia discrete graphics in a laptop with dual graphics. Windows by default decides which graphics to use and can switch at any time. Forcing Windows to always use the built-in Intel graphics solved the hanging problem. 

Last night Orca Slicer crashed and for the first time did not give me a chance to recover my work when I restarted it. That was disappointing because I had gotten used to having it crash without losing anything.

Just now, for the first time, Orca Slicer crashed while I was changing something. In the past it has always crashed when I didn't have it in the foreground using it. For some reason this reminded me about the graphics issue. I forced windows to always use Intel graphics for OrcaSlicer and if it doesn't crash again for a few days I will close this bug.

julie777, 2023-04-13:
This is the change that I made to force use of intel graphics.

[image]

julie777, 2023-04-13:
Here is a link to the Bambu Slicer issue which is long and has more detail.
#1361

julie777, 2023-04-15:
Sadly, I just had another crash of Orca Slicer. I guess the crash has a different cause.

julie777, 2023-04-16:
I am just adding more information.
Although the crash on the 14th happened, that is the only crash I have experienced since changing the graphics setting to only use Intel graphics. I have been running Orca Slicer 24x7 since I made the change and only that one crash, which is much different that previously when crashing multiple times a day.

The fact that the single crash after changing the graphics setting was during actual manipulation of objects on plates, could indicate that there is a different bug that caused it. 
Maybe the graphics change did fix the original crash.

I will keep updating this as new information surfaces.

SoftFever, 2023-04-19:
@julie777, thanks for the updates.
It looks like the trick works well

julie777, 2023-05-06:
I still get frequent crashes. The slicer logs just end so there is no info that can help.

I checked the Windows event log and I have five Orca crashes for just yesterday and today.
Scanning back through about a week of entries, almost all of them faulted in bambu_source.dll, although there were a couple in orcaslicer.dll.

@SoftFever 

I am still trying to get a usable build up and running. I don't completely understand how the build is partitioned into dlls. What does the orcaslicer.dll contain?

I realize that bambu_source.dll and bambu_networking.dll or downloaded at runtime.
I just upgraded bambu studio, and in the install directory it actually has a few pdb files, and one is for bambu_networking. Have you ever debugged into the networking dll?

closed 2023-11-20 by github-actions[bot]