OrcaSlicer/OrcaSlicer #723: can't get past filament select

thomllama, 2023-04-10:
**Describe the bug**
Setup locks at "select filament" .. there isn't anything to select but ALL  (Mac M1 version of app)

**To Reproduce**
try and start app, won't get past the setup wizard 

**Expected behavior**
see a list of filament?  I got no clue? 
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**
I had to select a rat rig as there is no place to select a custom setup 

**Desktop (please complete the following information):**
 - OS:  Mac mini M1  -  OS Ventura 
 - Version 13.2.1

thomllama, 2023-04-10:
fixed, guess you just need to restart the computer after install.. sorry to bother you!

hliebscher, 2023-04-11:
@thomllama  pls close the issue

closed 2023-04-11 by thomllama