OrcaSlicer/OrcaSlicer #727: Ender 5 Plus Config

SaturnsVoid, 2023-04-11:
Would be great if OrcaSlicer had a Ender 5 Plus profile, right now i am using SuperSlicer and the development has stagnated.

asoderberg, 2023-04-11:
I second this request. Thanks!

SaturnsVoid, 2023-04-12:
It would be great if there was a easy to use way to add custom printers to the configs.

hliebscher, 2023-04-14:
the easy way is take other ender printer and save as ender 5 plus and change the profile settings.
x y z and speed and accel 

that's it  

then it's easy to add your custom printer json file

SaturnsVoid, 2023-04-18:
I just took the time to take the stock Ender 3 profile and edit the settings to mach (at least the best i could) to my custom Ender 5+ settings from my SuperSlicer profile, Giving it a test now to see.

closed 2023-11-20 by github-actions[bot]