OrcaSlicer/OrcaSlicer #728: Fuzzy Font

MSBBZBL, 2023-04-11:
**Describe the bug**
All fonts are fuzzy and unsharp, If you dont use a standard Windows Font

**To Reproduce**
Use "Tahoma" as system font and don't use any font sharpening like "Clear Type"

**Screenshots**
https://i.imgur.com/uXiEELK.jpeg

In Cura, you can use your own font and font size -> https://gist.github.com/fieldOfView/7de3ad07fa06afb821bc0c5fe0b668dc

SoftFever, 2023-04-15:
The missing font causes it.
In the next version, the application should be able to load private fonts.
Before that, you can also manually install the fonts by yourself: just double-click the fonts files in `OrcaSlicer_install_folder`/resource/font`, then click install.

MSBBZBL, 2023-04-15:
I installed all six fonts, but the font is still the ugly mess.

MSBBZBL, 2023-04-28:
I checkt it again. Its "ClearType". Is ClearType disabled, the font is fuzzy, because this stupid Harmony font does only work with ClearType enabled....

Do we get systemfont in 1.6.3?

closed 2023-11-20 by github-actions[bot]