OrcaSlicer/OrcaSlicer #732: [Feature Request] Allow bed z-offset to perform boolean operations

H4kunaM4tata, 2023-04-12:
Currently, you can add negative blocks but that is all about it

[image]

If you slice it, things go sideways

[image]

The only way to get it to somewhat work is by adding a negative block for the first layers only.
Too much and it will fail.

[image]

With Cura, you can lower the model like so and anything before the bed surface is "removed"
This is useful for boolean operations like you wanna test a section of a model before sending hours long prints to find out the quality isn't great.

[image]

Cura automatically cuts/delete/boolean what is under the bed level making it so much easier to test a model section without using any CAD software just to perform that boolean operation or use another slicer for the matter.

[image]

Thank you

SoftFever, 2023-04-18:
For now, a workaround is to split the model and keep only the upper part.

H4kunaM4tata, 2023-04-18:
> For now, a workaround is to split the model and keep only the upper part.
> 

Never thought about it, thank you.