OrcaSlicer/OrcaSlicer #735: Models Appear solid black

K32design, 2023-04-12:
When loading a new .stl file the model appears on screen in a black colour and it is impossible to see any details of the model.

I have tried several different .stl files and models of varying geometry and complexity and get the same result

[image]

**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

Ibot-11, 2023-04-13:
Change the filament color
[image]
or
[image]

K32design, 2023-04-13:
I have tried this and it doesnt change anything.
tried several colours, tried selecting the part and then the filament colour, tried changing filament type and colour but its always black

[image]

closed 2023-04-13 by K32design