OrcaSlicer/OrcaSlicer #736: X-Y hole/contour compensation creates unexpected inner wall shrink

broxinagil, 2023-04-12:
Setting X-Y compensation (either hole or contour) creates unexpected shrink on inner walls on certain layers, where e.g. holes are in the model. This does not affect outer walls. This issue is as well seen on original Bambu Studio 1.5.0.61 and Softfever 1.4.4.

**with compensation:**
[image]

**without compensation:**
[image]

kevinakasam, 2023-04-13:
Bump on this, just got the same issue today. Used latest version 1.6.1

[image]
[image]
[image]

SoftFever, 2023-04-15:
Can you share a sample project?

broxinagil, 2023-04-15:
[LampWZ.zip](https://github.com/SoftFever/OrcaSlicer/files/11238735/LampWZ.zip)
Here it is

osmanoney, 2023-10-21:
X Y compensation should be used as an active value and there should not be that much protrusion. this is a big problem.stl file is attached

NexGen-3D-Printing, 2023-11-01:
Im also seing major issues with this feature, including the inner wall shinkage near holels, its cuasing all sorts of anomlies in prints, I have opted to disable it for now, using scale to compensate for material shrinkage might work better.

osmanoney, 2023-11-01:
prusa does not give such an error either. as a result, this is an error in orca and should be corrected

SoftFever, 2023-11-11:
When a vertical hole is present, the algorithm fails to detect horizontal holes because, technically, they are no longer considered holes.

PrusaSlicer doesn't have a hole compensation feature. Instead, its XY size compensation is equivalent to Orca's shrinkage feature. You can use this as a workaround.

[image]

[image]

closed 2023-11-19 by SoftFever

gbkwiatt, 2024-12-22:
@SoftFever Will this ever be fixed/corrected or accounted for with different feature, or  "it is what it is" ?