OrcaSlicer/OrcaSlicer #740: OrcaSlicer vs Superslicer printing speed and filament consumption

reapola, 2023-04-12:
**Describe the bug**
Wishing to move from Superslicer to OrcaSlicer and have copied my settings across but I'm seeing a difference in benchy printing time of 18mins along with ~25cm of filament used.  Any suggestions on what could be causing this?

**To Reproduce**
Take a Superslicer profile and replicate in OrcaSlicer using classic perimeters.

**Expected behavior**
Printing time and filament used should be similar.

More detail....

**Estimated time to complete**
[image]

**Printed time difference to confirm**
[image]

If I set small perimeters threshold to 0mm instead of max of 20mm set in SS then I can get the time down to a difference of 7mins

**Orca Setting**
[image]

**SS setting**
[image]

**Speed comparison with small perimeters set to 0 on orca.**
[image]

**Time comparison with small perimeters set to 0 on orca.**
[image]

**Printer model**
Voron 2.4 300

**Desktop (please complete the following information):**
 - OS: Windows 11 with Orca 1.6.1

reapola, 2023-04-12:
**Superslicer settings (rename to .ini)**
[SuperSlicer_config_bundle.txt](https://github.com/SoftFever/OrcaSlicer/files/11210916/SuperSlicer_config_bundle.txt)

**Orca Settings (rename to .json)**
[Metis Voron 2.4 300 0.4 nozzle.txt](https://github.com/SoftFever/OrcaSlicer/files/11210930/Metis.Voron.2.4.300.0.4.nozzle.txt)
[Voron eSUN ABS.txt](https://github.com/SoftFever/OrcaSlicer/files/11210931/Voron.eSUN.ABS.txt)
[Metis 0.20mm Standard @Voron.txt](https://github.com/SoftFever/OrcaSlicer/files/11210932/Metis.0.20mm.Standard.%40Voron.txt)

H4kunaM4tata, 2023-04-15:
I don't think you will have them both printing exactly with the same time.

Trident here with `klipper_estimator`, in many cases OrcaSlicer is actually faster than PrusaSlicer, in others I've found that OrcaSlicer was taking longer because the profiles weren't exactly the same.

SoftFever, 2023-04-18:
Thanks. 
Closing this ticket as this is not a bug.

commit ea5c88e3 (2022-12-13): merge upstream changes (#25)

* Add sigxcpu76's fix to force use of light GTK theme

When using Bambu Studio with a dark themed Linux desktop, many of the dialogs appear as white text on a white background (font's correctly invert, but dialogs remain white).

@sigxcpu76 provided a workaround for this in #12 which is to tell the app to use a light GTK theme.  This change incorporates that workaround into the AppImage making the app more usable until such time as proper dark theme support can be added to Bambu Studio.

* Update Containerfile to make it Docker compatible

- Replace COPY command w [... cut]
commit 7e96cbf7 (2022-12-13): Fixes #740

Removing check for local or Remote FX virtualized RDP session. The checks below this line already verify if proper OpenGL support is present.
closed 2023-04-18 by SoftFever