OrcaSlicer/OrcaSlicer #741: Nozzle scratch top layer/surface

rcxrbt, 2023-04-12:
**Describe the bug**
I'm new in OrcaSLicer and I'm not sure if this is a bug or I'm wrong, but whit other slicer,  I don't have it.
Nozzle scratch top layer/surface on travel... please, see the pic.

**3mf File for This Bug**
[nozzle_scracth.zip](https://github.com/SoftFever/OrcaSlicer/files/11215049/nozzle_scracth.zip)

**To Reproduce**
Normale slicing

**Expected behavior**
Add some behavior to lift Z axis during travel on top layer/surface or not travl on surface (like Cura)
[image]

**Screenshots**
[image]

**Printer model**
N/A

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.6.1

Tnx!

hliebscher, 2023-04-14:
@rcxrbt  in printer settings you have z-hop = 0 set to 0,4

rcxrbt, 2023-04-14:
> 

ok, but this is always involved? like every retraction or only in top surface?

rcxrbt, 2023-04-17:
> @rcxrbt in printer settings you have z-hop = 0 set to 0,4

Do you refer this?

[image]

I dont want to z-hop every retraction...

closed 2023-11-20 by github-actions[bot]