OrcaSlicer/OrcaSlicer #742: Project File doesn’t load properly

Ibot-11, 2023-04-12:
When I open a 3MF project Orcaslicer is not able to load it properly. Everything looks fine until I save the file and restart / reopen the project. 
Some models have very poor mesh quality and many multicolor prints are shifted. When exporting them as a generic 3mf everything works fine.

[A_Drive_Lower.zip](https://github.com/SoftFever/OrcaSlicer/files/11215246/A_Drive_Lower.zip)
[image]

SoftFever, 2023-04-13:
Seems I can't download the file

Ibot-11, 2023-04-13:
Okay for me the download works.
Here is a reupload:
[A_Drive_Lower.zip](https://github.com/SoftFever/OrcaSlicer/files/11225494/A_Drive_Lower.zip)
You have to unzip the file since I can’t upload a 3MF file to github.

hliebscher, 2023-04-14:
sometimes i have the same issue if the stl's are very big ... an many plates 
Mac M1  but it is since V 1.5 but not to reproduce

closed 2023-11-20 by github-actions[bot]