OrcaSlicer/OrcaSlicer #743: AMS Filament and user profiles have separate color now

DuckTown407, 2023-04-12:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/686

<div type='discussions-op-text'>

<sup>Originally posted by **DuckTown407** April  6, 2023</sup>
In 1.5 when you had the AMS sync you would get the colors and filament types that you did on the AMS available in the slicer. Mine would always be a generic because I use brands that BL doesn't have profiles for. I would then set each one to a specific user profile that contains the real filament settings and the color would stay what I had set in the AMS. Now, in 1.6, all non AMS filaments  have a "default color" such that when you change an AMS filament to this type it overrides the color. Say I loaded in 2 PLA spools from the same brand and wanted  to use  those for a multicolor print: now I would need to have 2 different profiles, 1 for each color of that filament, rather than just selecting my user defined profile and assigning it to each AMS slot. Is this the intended behavior? Can there be a setting to make it the color of the AMS slot you chose? </div>

Photogad, 2023-04-13:
https://github.com/SoftFever/OrcaSlicer/issues/651

SoftFever, 2023-04-13:
dup with https://github.com/SoftFever/OrcaSlicer/issues/651