OrcaSlicer/OrcaSlicer #745: Sandwich wall mode wrong wall order when neighbouring features spaced less than one internal line width apart (Benchy)

igiannakas, 2023-04-13:
**Describe the bug**
Layer 65 on the attached benchy (below the chimney, where the rear lettering is) results in the wrong printing order when inner-outer-inner mode is selected. The Inner most line is printed, then the internal perimeter, then the external perimeter. I suspect the slicer considers the inner most perimeter as a feature linked to the chimney stack and prints it as part of that instead of the outer layer line resulting in the wrong order.

That ordering issue is causing a "bulge" on the surface of the model.

**3mf File for This Bug**
[3DBenchy.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11219527/3DBenchy.3mf.zip)

**To Reproduce**
Slice using inner-outer-inner mode. Set layer height to 0.1 (to cross check the layer number).

**Expected behavior**
The external perimeter is printed after the inner most perimeter irrespective of other features.

**Screenshots**
[image]

If applicable, add screenshots to help explain your problem.

**Printer model**
Ender 3 S1 Pro

**Desktop (please complete the following information):**
 - OS: OSX M1 Pro
 - Version Ventura 13.2.1

SoftFever, 2023-04-13:
> I suspect the slicer considers the second internal perimeter as a feature linked to the chimney perimeter and prints it as part of that instead of the outer layer line resulting in the wrong order because the chimney stack gets printed first.

This is a correct speculation. 
Sandwich mode, like the outer wall first mode, is mainly designed for mechanical parts that have simple shapes and no large overhangs. 
While it can be optimized, it is a rare use case and therefore has low priority.

igiannakas, 2023-04-13:
> 

Agreed, I've reviewed the code doing the wall ordering and it appears that a tweaked planning algorithm is needed for this.

In my view, this method of printing should be the standard, it gives fantastic quality exterior surfaces without the downside of a potentially ugly seam when traveling over a long distance from the previous layer or infill. Personally I've found it fantastically useful for all prints :) 

being a developer myself if I manage to wrap my head around the code structure I may give it a try :)

igiannakas, 2023-04-13:
One thought that might solve this and would love your thoughts.

Looking at the PerimeterGenerator.cpp it appears to me that we are using the internal to external wall ordering as a baseline and swapping out the last internal perimeter with the external in the sequence. 
So imagine lines 1-2-3 (outside in), printed in this order 3-1-2.
But if there is no perimeter 3, it would fall back to printing inside out (2-1).

That gives weight to the inner most perimeter being printed first and then the external.

`   if (this->config->wall_infill_order == WallInfillOrder::InnerOuterInnerInfill)
            if (ordered_extrusions.size() > 1) {
                int last_outer = 0;
                int outer      = 0;
                for (; outer < ordered_extrusions.size(); ++outer)
                    if (ordered_extrusions[outer].extrusion->inset_idx == 0 && outer - last_outer > 1) {
                        std::swap(ordered_extrusions[outer], ordered_extrusions[outer - 1]);
                        last_outer = outer;
                    }
            }
`

What if we biased the algorithm to be the other way around - weigh the outside perimeter first in the scenario of 2 wall loops - this could be done like this:

Use the external perimeter wall ordering as the baseline and re-ordered based on that. Imagine lines 1-2-3 (outside in) and swap line 3 with line 1 in print order. So it would be printed like: 3-1-2. If there are only two perimeters the reorder loop wouldn't run as we need to test for ordered_extrusions.size() > 2 so it would print the external perimeter first then the internal.

It would result in a slightly ugly seam but would retain surface quality across the remaining surface for areas where only 2 extrusions exist.

I'm really struggling to compile Orca on my Mac to try it out :( 

Code change could look like the below:

`bool is_outer_wall_first =
                this->config->wall_infill_order == WallInfillOrder::OuterInnerInfill ||
                this->config->wall_infill_order == WallInfillOrder::InfillOuterInner||
               this->config->wall_infill_order == WallInfillOrder::InnerOuterInnerInfill
`

`   if (this->config->wall_infill_order == WallInfillOrder::InnerOuterInnerInfill)
            if (ordered_extrusions.size() > 2) {
                std::swap(ordered_extrusions[2], ordered_extrusions[0]);
            }
`
As a bonus you dont need to iterate through all the remaining extrusion lines - just reorder the first and third, then the second, fourth, fifth etc would print in order. 

What do you think?

SoftFever, 2023-04-14:
@igiannakas Great. Feel free to create a PR when you feel comfortable.

igiannakas, 2023-04-14:
> @igiannakas Great. Feel free to create a PR when you feel comfortable.

Will do, love your work by the way, even though I don’t own a Bambu lab this slicer is fantastic! 👏🏻🙏🏻

closed 2023-09-18 by igiannakas