OrcaSlicer/OrcaSlicer #748: Objects and plate positioning broken when using old .3mf project

VDechev, 2023-04-13:
When I'm trying to open older project which has multiple plates with the new version the positioning is not working properly. Even if I add a new plate, which is repositioning the plates the space between them is not the same as older version. This happens to Bambu Lab and third party printers, In my case I have more than 100 files which are adjusted for 4 different printers. Every file has multiple plates(up to 20) . 

I'm not able to provide .3mf file directly(the files are from a client), but I would try to reproduce it with free models and upload it if needed

To reproduce it I guess you need to create a project with older version(1.4.5 or older) with multiple plates and then open it with newer version(in my case 1.6.1)

In the screenshots below the files are covered in squares, but they represent the size of the actual models

This is how the project looks like in 1.4.5 
[image]

This is one of the cases when opening the project in 1.6.1(plates are squished). If I add a new empty plate the case would look like the next example(too much space between the plates)
[image]

This is the second case when opening the project in 1.6.1(plates are far away from each other)
[image]

It happens on all of my printers. They are Voron 2.4, Voron Trident, P1P, X1C

I'm using Windows OS

SoftFever, 2023-04-14:
@VDechev thanks.
A sample project will be helpful.

austinrdennis, 2023-11-11:
@SoftFever @VDechev I think I have more info on this, but maybe my issue is different. It only happens on the second row of plates and subsequent rows. It also seems to happen specifically when I'm working on a 3mf and switch to another 3mf without closing the app. The newly opened 3mf has what I'm guessing is the old position data from the 3mf I just closed as I notice it when I switch from my 300x300 printer to my 235x235 printer.

Here's what it looks like when you open the file. The first (top) row of plates is correct but the next rows are all offset of center by the same offset it looks like.
[image]

[Here's](https://drive.google.com/drive/folders/1VHBxCU9uU6loDuE4j7gFqt4OGCncIDXj?usp=drive_link) my 3mf files so you can try it.

I'm on windows and it seems to be in a lot of older versions but I'm specifically on 1.8.0-RC. I think it happens on my Mac (M series) as well, just haven't tested it thoroughly. Let me know what could help you more in diagnosing the issue.

SoftFever, 2023-12-30:
Some updates on this matter:
The issue occurs when you have a different printer model (with a different bed size) activated in OrcaSlicer while opening a 3mf project that uses another printer model.
The current workaround is to manually switch to the corresponding printer before opening the project.
I will be addressing this issue in the next release.
For now, please use the suggested workaround.

igiannakas, 2024-01-02:
Adding project 2x project files with similar issues in opening. Always opening on the same printer & plate combo so the workaround here doesnt seem to work.

[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/13808931/Archive.zip)

KevAquila, 2024-07-05:
Just chiming in and confirming that the "workaround" does not work anymore.  This makes it rather difficult to share large project files with people.

Felix14-v2, 2026-10-05:
> Always opening on the same printer & plate combo so the workaround here doesnt seem to work.
> 
> [Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/13808931/Archive.zip)

Unable to reproduce in latest 2.5.0-nightly (eb5b9a77)

[image]

closed 2026-10-07 by ianalexis