OrcaSlicer/OrcaSlicer #750: LINK : fatal error LNK1181: "D:\a\OrcaSlicer\OrcaSlicer\deps\build\OrcaSlicer_dep\usr\local\lib\freetype.lib"

hliebscher, 2023-04-14:
in windows build is a path bug also in the prebuild OrcaSlicer_dep_10April2023

wrong location 
"D:\a\OrcaSlicer\OrcaSlicer\deps\build\OrcaSlicer_dep\usr\local\lib\freetype.lib"

it has to be 
"\OrcaSlicer\deps\build\OrcaSlicer_dep\usr\local\lib\freetype.lib"

Workaround copy freetype.lib to D:/a/Orca.......

it is only in windows build  mac works fine
it seems to be hardcodet  i have no d: drive 
i found some d: path in source code .... made PR

SoftFever, 2023-04-14:
You will need to build the deps on your machine by yourself.
Currently the absolute path is used.

If you want to use the prebuilt OrcaSlicer_dep_10April2023 binaries, you need to set the path same, it is built for CICD

SoftFever, 2023-04-14:
run `build_release.bat deps` in Visual studio 2019 command line to build deps

hliebscher, 2023-04-14:
then i get a boost build hash error ....
now its running thanks

closed 2023-04-14 by hliebscher

hliebscher, 2023-04-14:
@SoftFever i change the batch script then it works
DCMAKE_BUILD_TYPE=Release -DWIN10SDK_PATH="C:/Program Files (x86)/Windows Kits/10"
it it the same path like windows registry