OrcaSlicer/OrcaSlicer #752: Remove limit on nozzle size

gaiosa, 2023-04-14:
Currently, the nozzle diameter size is limited to a maximum of 1.0 mm. Sometimes I want to use a nozzle diameter that is larger than 1.0 mm, so I can make strong parts with large layer widths and heights. However, if I am using a 1.4 mm nozzle, I have to set the machine to a 1.0 mm nozzle. This means I can't set the layer height any higher than the nozzle size of 1.0 mm, but I like to run the 1.4 mm nozzle with a layer height of 1.12 mm (80% of nozzle size). Could the maximum nozzle size limit be removed?

[image]

SoftFever, 2023-04-14:
will relax the limitation in next build

gaiosa, 2023-04-17:
Great, thanks!

closed 2023-04-17 by gaiosa

FlowForm, 2023-04-18:
Awesome, gonna test my 1.8 very soon :-)

SoftFever, 2023-04-19:
1.8mm nozzle go go go!
:)

[image]

MoiRam14, 2024-10-31:
I was able to print with 1.4mm nozzle using custom preset I made like 2 or 3 months ago after I upgraded to the biqu panda hottend and bought some revo nozzles. But now I see that I get stuck on the nozzle not matching the size set on printer parts setting. Can't add any new sizes to that list. So effectively I can't use my 1mm, 1.2mm or 1.4mm nozzle anymore.