OrcaSlicer/OrcaSlicer #755: Slow filament change display after failover.

vicpylon, 2023-04-14:
**Describe the bug**
When the printer fails over to another roll of identical filament when the first roll empties, the display takes 10-15 minutes to update to show it has switched to the new roll.

**To Reproduce**
Let the machine fail over and watch the status screen. This took about 15 minutes to appear. It started on the third roll and then eventually showed the fourth roll being  used.

[image]

**Expected behavior**
Should change the display much faster, since a manual check of the printer indicated it was using the new roll. 
**Screenshots**

**Printer model**

**Desktop (please complete the following information):**
Mac M1 
Ventura 13.2.1

SoftFever, 2023-04-14:
Thank you for submitting the ticket. 
However, since this issue requires interaction with Bambu's proprietary firmware, it would be best to submit a request on their repository.

SoftFever, 2023-04-14:
Closing it as explained.

closed 2023-04-14 by SoftFever