OrcaSlicer/OrcaSlicer #757: Incorrect display of Cyrillic fonts when trying to translate (Russian, Ukrainian, Bulgarian)

Andrewvs599, 2023-04-15:
**Describe the bug**
When trying to create a translation of the program interface into Russian, Ukrainian or Bulgarian, Cyrillic characters are displayed as ?????? in some application areas. The main interface is displayed correctly. There is also a problem with Cyrillic characters described in #671.

**To Reproduce**
Create a translation file for the specified languages and replace the translation file of any language in the i18n folder with it. Select the replaced language in the program properties.

**expected behavior.**
Correct display of Cyrillic

**Printer model**
Bambu Lab X1 Carbon Combo

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 21H2
 
[image]
[image]
[image]

SoftFever, 2023-04-16:
@Andrewvs599 
Can you create a PR for this so that I can take a look what could be wrong?

gogsi1, 2023-06-23:
> @Andrewvs599 Can you create a PR for this so that I can take a look what could be wrong?

Please write if you are going to fix this problem?

closed 2023-11-20 by github-actions[bot]