OrcaSlicer/OrcaSlicer #769: Flow Calibration Setting

jagr200, 2023-04-17:
I would LOVE it if "flow calibration" could be toggled in settings. So we can set it to default to off and we have to manually check it. If I have done all of the calibrations I do not want to run flow calibration. I would also love the same for "Bed Leveling" as I only run it when I have powered my machines down.

Thank you!

SoftFever, 2023-04-18:
OS will remember your last choice so that basically covers your use case.

closed 2023-04-18 by SoftFever

jagr200, 2023-04-18:
> OS will remember your last choice so that basically covers your use case.

How is this complete? When I go send it to my machine I have to toggle the setting every single time. Would a toggle in preferences be too hard to setup so we don’t have to toggle that every time?

SoftFever, 2023-04-18:
OS should remember your last toggle state of these settings.
If that doesn’t work on your machine, a video to show how to repro the issue will be helpful

jagr200, 2023-04-18:
> OS should remember your last toggle state of these settings.
> If that doesn’t work on your machine, a video to show how to repro the issue will be helpful

The video wouldn’t show anything. 

I slice the plate. I hit send to printer. All 4 boxes are checked. Leveling, flow calibration, Timelapse and enable AMS. 

I want to be able to go into preferences and turn off leveling and flow permanently.