OrcaSlicer/OrcaSlicer #776: Option in Print Submitting to send data variables to Print_Start Macro

behold81, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
not always wanting to run a Bed Mesh and to support possible future preprint calibrations

**Describe the solution you'd like**
similar to printing on a BBL Printer you can turn off some prechecks. you select klipper supported functions that will send a variable usable in a Start_Print to disable or enable a function. 

**Describe alternatives you've considered**
buying a BBL X1C :D

**Additional context**
Add any other context or screenshots about the feature request here.

hliebscher, 2023-04-18:
@behold81    the easy way is for you copy the printer profile without the klipper start macro and take only the G-code that you need in this config and a second profile with more or less....

behold81, 2023-04-18:
Agree that's an option. But why not make it more integrated like the bbl setup where on sending a print you can choose the settings you want?

hliebscher, 2023-04-18:
OrcaSlicer is diferent for BBL and other Printer and your feature is only for klipper but is possible

behold81, 2023-04-18:
Exactly my thought. If printer is either configured for Klipper, or if you enable against the printer profile then you will get the popup to choose your options for printing. I would say long term this feature would be great in Klipper itself too for local printing without the slicer.

SoftFever, 2023-04-18:
With Klipper you have much better solutions than BBL like [KAMP](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging)

closed 2023-04-18 by SoftFever