OrcaSlicer/OrcaSlicer #777: running v1.6.2-beta creates a notification to download v1.6.2.beta

julie777, 2023-04-17:
**Describe the bug**

I unzip and run v1.6.2-beta. It opens a notification dialog.

[image]

I don't think a notification should appear unless there is a newer version than the one running.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

tsmith35, 2023-04-17:
Seeing the same here. I selected the box to not remind me about this version again, but perhaps it's an easy fix. :-)

SoftFever, 2023-04-18:
Yeah, a wrong tag name was created previously.
It should have been fixed now.
Thanks for reporting it.

closed 2023-04-18 by SoftFever