OrcaSlicer/OrcaSlicer #781: V400 Printer bug

wagzledz, 2023-04-18:
**Describe the bug**
Thumbnail isn't generated

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Send a sliced Gcode to the V400, you'll notice the thumbnail isn't present 

**Expected behavior**
A thumbnail is generated

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
FLSun V400
**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] iOS 
 - Version [e.g. 22]

hliebscher, 2023-04-18:
you need a newer Klipper version ! 
NOTE:
1. Please check [here ](https://github.com/SoftFever/OrcaSlicer#how-to-install)for installation instructions.
2. If your profiles didn't show up after updating, please restart OrcaSlicer
3. Klipper users please update Moonraker to the latest version so that it can recognize OrcaSlicer and parse metadata like thumbnails.
**4. If you're running Klipper, it's recommended to add the following configuration to your printer.cfg file.

# Enable object exclusion
[exclude_object]

# Enable arcs support
[gcode_arcs]
resolution: 0.1

SoftFever, 2023-04-18:
Hi @wagzledz,
As @hliebscher mentioned, it requires latest moonraker to recognize OrcaSlicer properly.

wagzledz, 2023-04-18:
Yes I will have to attempt to update klipper.  Not sure the stock FLSun v400 pad will do that?  I will try.  Thank you.

SoftFever, 2023-04-18:
Closing it since it's a bug.

closed 2023-04-18 by SoftFever