OrcaSlicer/OrcaSlicer #783: Outer wall line width not changing

TechDude23, 2023-04-18:
I have been testing the line width setting and the outer wall will not adjust it will always be @ .4 the fist layer line width is working ok.

So I did some custom g code for another slicer and I can change the line width so I know its not the Bambu Lab hardware.

Orca Slicer V 1.6.1 

Bambu Lab X1

Window OS: 10

Thank you for fixing the export bug :)

SoftFever, 2023-04-18:
[image]
@TechDude23 Works fine in 1.6.2
Please attach a sample project if you still find the issue in 1.6.2

TechDude23, 2023-04-18:
Just tested it out with version 1.6.2 and it is fixed woohoo :)

Thank you so much

closed 2023-04-18 by TechDude23