OrcaSlicer/OrcaSlicer #784: Two feature request

TechDude23, 2023-04-18:
It would be cool to have in the filament settings a ( note tab ) so you can keep typed text notes.

Another cool thing would be when you change filament colors the flow ratio and pressure advance setting plus nozzle temps would change to your settings for that filament color or maybe have a drop down menu to select the color profile.

hliebscher, 2023-04-18:
for the first you can write text notes in gcode 
for the second point i copy the filament setting for each color and change in this settings the default color , PA, temp
so i have a drop down

hliebscher, 2023-04-18:
[image]

TechDude23, 2023-04-18:
Hi Hliebscher,

That's what I have been doing for now it just wanting to keep the G code clean looking from my notes plus I have over 150 types and colors of filaments and the filament list is getting so long.

So now I only keep 4 filaments of the same brand and type and will just change colors and input the settings from my notes for the color in use. 

Thank you for posting your good tips it will help someone out :)

closed 2023-11-20 by github-actions[bot]