OrcaSlicer/OrcaSlicer #785: Multi Coloring does not work correctly

ccomtest, 2023-04-18:
**Describe the bug**
When painting a model with a color, it uses first color often/does not fill complete wall/object.
Print doesn't look as expected in full color.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
[EMPACT_Logo_Color_Issue.zip](https://github.com/SoftFever/OrcaSlicer/files/11260496/EMPACT_Logo_Color_Issue.zip)

**To Reproduce**
Color a model (text) with more than one color. When you slice it, you see your set color is not fully painted and will also print like this.

**Expected behavior**
When i color a wall, it should print the whole wall in this color. I am not sure, the Fusion Modell has an impact on this. It seems, there a lot of shapes in a Text modell. But when i incease fill angle to possible maximum it also does not color the whole wall, which should be not the case. Also the painting looks like, it has painted all. When sliced, it shows how the print really will look like.
**Screenshots**
[image]
[image]
[image]

**Printer model**
Bambu Lab X1C

**Desktop (please complete the following information):**
 - Windows 10, Orca Slicer 1.6.1

ccomtest, 2023-04-18:
See also "one layer deep coloring only". Which descibes my issue a bit better i think as an inssue in Bambu Slicer.
Coloring does not behave like in Super Slicer.

ccomtest, 2023-04-18:
It seems there are more bugs according to color printing (no one else has these issues?).

When i set a top layer in the middle for color difference, it seems the color underneath is also changed, but not for walls.
[image]
[image]

Leasing into more color changes and wrong prints.

SoftFever, 2023-04-18:
Thanks @ccomtest .
In that case, hopefully they may have been already working on this.
Closing it.

closed 2023-04-18 by SoftFever

FloodedZebra0, 2024-08-03:
I am having this same problem as well, has anyone or orca/bambu slicer fixed this yet?