OrcaSlicer/OrcaSlicer #790: Some parts not receiving "top layer" processing.

chmarr, 2023-04-19:
**Describe the bug**
With a particular STL, the top layer of certain sections are terminating with the normal "inner wall" rather than "top layer infill". Note that this problem does _not_ occur with Bambu Studio 1.5.0.60

**3mf File for This Bug**
3mf file attached.
[chainmail_flowalistik_sample.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11268014/chainmail_flowalistik_sample.3mf.zip)

**To Reproduce**
* Load 3MF
* Use the standard "0.20mm, Standard" setting. (albeit problem occurs at most settings)
* Slice
* Observe that about half the components have their inner walls showing, rather than "top surface".

**Expected behavior**
The tops of all components show the red "Top Surface" indication.

**Screenshots**
Screenshot attached.
[image]

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: MacOS 13.3.1
 - OrcaSlicer Version: 1.6.1

chmarr, 2023-04-19:
A quick addition. The parts that aren't receiving the Top Layer processing are indeed shorter than the parts that are. That shouldn't stop them receiving the processing, I would think.

defcrash, 2023-04-25:
I have the same problem, if i fill a model with only walls i get no top layers:
[image]

Now sliced with only 3 walls (it looks better, it seems if a specific part only have walls it dosent add top layers):
[image]

This only happens with Arachne enabled, if i switch to Classic mode it will slice correctly:
[image]

To0wnn, 2023-07-16:
Hi,

This issue is still there in classic mode. The toplayer is not looking nice when printing
[image]
With superslicer it is ok but maybe that is just an oppinion.
[image]
Hoopfully it can be fixed in the new release or did i miss a setting?

closed 2023-11-20 by github-actions[bot]