OrcaSlicer/OrcaSlicer #791: Device tab empty if using printers hostname instead of the ip address

ffavero, 2023-04-19:
The device tab doesn't show anything if in the physical printer configuration the hostname of the printer is used instead of the ip address (in my case device ip can change, I base assignments on hostnames).
Using the hostname to connect with the printer works out of the box to upload gcodes.

To reproduce just replace the printer ip with the hostname, chack if the test passes correctly (it should), and try to upload a file to the printer (it should). Device tab is empty though

**Screenshots**

With hostname

[image]
[image]

With IP
[image]

[image]

**Printer model**
ZeroG Mercury One.1

**Desktop (please complete the following information):**
macOS silicon M1

fire1ce, 2023-04-19:
@ffavero can confurme same behavior.

SoftFever, 2023-04-20:
@fire1ce what OS do you use?

fire1ce, 2023-04-20:
@SoftFever macOS silicon M1

gargamel778, 2023-04-21:
Can confirm similar behavior on x86 MacOS version

ammmze, 2023-05-12:
I can confirm the same thing. However, I did find that if I run it through a TLS terminating proxy ( just tested with ngrok), giving me an http**s** connection instead of http, then it would work.

SoftFever, 2023-05-13:
I've seen similar issues on Windows when I port printer network support from PrusaSlicer.
Later I found an solution/workaround in PS for Windows 10/11.
Might be the same cause.

> #ifdef WIN32
>     // Workaround for Windows 10/11 mDNS resolve issue, where two mDNS resolves in succession fail.
>     if (m_host.find("https://") == 0 || test_msg_or_host_ip.empty() || !GUI::get_app_config()->get_bool("allow_ip_resolve"))
> #endif // _WIN32
>     {
>         // If https is entered we assume signed ceritificate is being used
>         // IP resolving will not happen - it could resolve into address not being specified in cert
>         url = make_url("api/files/local");
>     }
> #ifdef WIN32
>     else {
>         // Workaround for Windows 10/11 mDNS resolve issue, where two mDNS resolves in succession fail.
>         // Curl uses easy_getinfo to get ip address of last successful transaction.
>         // If it got the address use it instead of the stored in "host" variable.
>         // This new address returns in "test_msg_or_host_ip" variable.
>         // Solves troubles of uploades failing with name address.
>         // in original address (m_host) replace host for resolved ip 
>         info_fn(L"resolve", test_msg_or_host_ip);
>         url = substitute_host(make_url("api/files/local"), GUI::into_u8(test_msg_or_host_ip));
>         BOOST_LOG_TRIVIAL(info) << "Upload address after ip resolve: " << url;
>     }
> #endif // _WIN32

ammmze, 2023-05-28:
I've got a fix for this coming soon...just doing some final testing then will open a PR.

ammmze, 2023-05-28:
PR is up now #1157

ammmze, 2023-05-28:
I'm still kind of on the fence whether we should disable ATS as discussed in the PR, but for those looking for a workaround to get this working today with your existing version, you can add a security exception for your domain like this. Open up a terminal window and run the following command, but replace "printer.lan" with the hostname of your printer. (And if the path to OrcaSlicer.app is different, change that as well)

```shell
plutil -replace NSAppTransportSecurity -json '{"NSExceptionDomains": {"printer.lan": {"NSExceptionAllowsInsecureHTTPLoads": true}}}' /Applications/OrcaSlicer.app/Contents/Info.plist
```

commit c675d979 (2023-05-31): fix(macos): disable App Transport Security (#1157)

* feat(webview): enable context menu

This enables the ability (at least on MacOS) to reload the webview.

* fix(webview): add handler for webview error

Adding this un-covered the following error when trying to load a the printer page using http and hostname:
```
OnError: error loading page about:blank  wxWEBVIEW_NAV_ERR_OTHER The resource could not be loaded because the App Transport Security policy requires the use of a secure connection.

```

* fix(macos): disable App Transport Security policy

Disables the [App Transport Security](https [... cut]
closed 2023-05-31 by SoftFever