OrcaSlicer/OrcaSlicer #796: NUMBER OF SLOW LAYERS in the beginning of the print.

RudiKleinSA, 2023-04-19:
Please add NUMBER OF SLOW layers in the beginning of the print.

This is vital for ABS not to Warp. 
I print x10 slow layers.

SLOW INITIAL LAYERS guarantees my print not to warp.

Thank You

SoftFever, 2023-05-21:
Added in 1.6.3 beta

closed 2023-05-21 by SoftFever

RudiKleinSA, 2023-05-21:
SoftFever,

Thank you very much.

I am using Windows - How can I get 1.6.3 beta?  
Just wait until it is released?

SoftFever, 2023-05-21:
@RudiKleinSA 
You can download the 1.6.3 beta release here: https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.3-beta

RudiKleinSA, 2023-05-21:
[image]

SoftFever, 

Thank you for replying.

I downloaded the new 1.6.3 beta release, and WAU, amazing. 

Thank you very much!!

I love ORCA Slicer.

Well done.