OrcaSlicer/OrcaSlicer #797: Jerk is not calculated in printtime

foxmckraut, 2023-04-19:
Model printing time is not including jerk
on 1 jerk its the same as on 1000 jerk.
[image]
[image]

1000 jerk:
[image]
[image]

Tried the same on Superslicer and works there.

SoftFever, 2023-04-20:
What printer/gcode-favor is this?

SoftFever, 2023-04-20:
Do take note though, I cap the max jerk value based on machine limits.
If your machine limits is set to 9, then it will cap to 9 even you set to 999 in speed settings.

foxmckraut, 2023-04-20:
that makes sense thanks for the explanation

SoftFever, 2023-04-21:
It looks like this has been resolved.
Feel free to re-open it if it's still doesn't work

closed 2023-04-21 by SoftFever