OrcaSlicer/OrcaSlicer #804: Klipper "timer too close" error on specific model, but Cura works fine ..

piotrb, 2023-04-20:
**Describe the bug**
I really don't have an amazing one .. this may take some back and fourth to get to the bottom of .. 

Both Orca 1.5 and 1.6.2. .. and SuperSlicer have this same issue .. (I didn't try Prusa to mix things up) .. 

The best I can describe it is that somehow .. after printing the first few mm (this seems to vary between ~2.5 to ~4.5) .. klipper blows up with the following message:

```
MCU 'toolhead' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

(and yes I'm running a multi mcu setup with a CANbus toolhead .. this shouldn't be a problem and isn't for other models .. )

I started  a bit of a monologue on the klipper discord logging my attempts .. 
https://discord.com/channels/431557959978450984/550725315278929941/1097022117603717141

all I can guess is somehow the slicer is generating gcode in such a way that its overwhemling klipper and it feels like its not keeping up .. (the error message is quite generic which really doesn't help)

The fun part is that when I slice the same model with Cura .. it works fine .. 

**3mf File for This Bug**
[stealthburner_printhead_revo_voron_front.zip](https://github.com/SoftFever/OrcaSlicer/files/11287926/stealthburner_printhead_revo_voron_front.zip)
(it wouldn't let me upload the 3mf directly, so I zipped it)

**To Reproduce**
eh .. print it .. and at least on my printer .. after ~2.5 - 4.5 mm it tends to kill the printer .. 

**Expected behavior**
it should print :)

**Screenshots**
n/a

**Printer model**
Voron Switchwire

**Desktop (please complete the following information):**
 - OS: Windows 11
 - 1.5.0, 1.6.2 (and also SuperSlicer b8e2779 since I had that sitting around for a test) .. but it does imply the issue is with the shared prusa codebase perhaps

Misterke, 2023-05-29:
I something happening on this one?  I'm starting to thing OrcaSlicer 1.6.2 is triggering something similar on my Klipperized Sapphire Plus.  Too early to state that as a fact though, but I have hit 3x "timer too close" now on a file sliced with OrcaSlicer 1.6.2.

piotrb, 2023-06-01:
I was relying on MVS to manage the speed for me .. but it seems in some sections where it was bursting past 100 it was too much .. I recently got the issue again when I was tweaking the speed settings and bumped infill to like 125 or 150 .. so I set it back to 100 and its happy again

Misterke, 2023-06-04:
But 100mm/s is peanuts and Klipper should not have any problems with that.  I'm printing infill at 450mm/s and movements up to 600mm/s and except for that one model with OrcaSlicer 1.6.3, that hasn't hit a "timer too close".  Looking at the load graphs after printing, I typically see bandwidth and load being way below 50%, but in that failing print at the failure point there was every time a sudden spike of bandwidth above 80%.

As this occurred at least 2 times (the 3rd one didn't fail at that point and I don't know whether there the spike was present) at the same point in the file, I'm guessing something in the gcode is causing that spike in bandwidth - possibly some way too detailed movements causing lots of commands over very short period of time.

I would expect that something like that could be detected with some basic gcode analysis before starting the print, no?  So, is there any tool that can give an insight into the bandwidth that would be used by a gcode file before actually having printed it and then analyzing the klippy.log?  Maybe @SoftFever knows whether something like that exists?

Note that in my case, just reslicing the file with the object positioned slightly differently seems to have avoided the problem or at least reduced the spike to levels where it doesn't fail anymore.  I'm unsure why that is though ...

piotrb, 2023-06-21:
yea its super weird .. not actually Orca's fault per say .. seems to be something in the PrussaSlicer base .. but yea its super weird .. its specific models .. and like you mentioned other conditions like even the positioning on the plate can make a difference ..

SoftFever, 2023-07-16:
@piotrb @Misterke 

I suspect the issue is caused by frequent fan speed changes. Orca and SuperSlicer may be adjusting the fan speed frequently for areas that contain many tiny overhangs and bridges. In contrast, Cura does not adjust the fan speed for overhangs, if I recall correctly, so it doesn't have this issue.

Can you confirm if this is the case on your side?

Meanwhile, I'm going to reach out to the Klipper community to inquire if Klipper can handle frequent M106 commands.

piotrb, 2023-07-16:
Cura was also substantially slower by default so that may have been a big part of it too .. not sure

Dustmuffins, 2023-07-24:
I was having a similar issue with this. It would stop at the same time in the print each time I tried it. Slicing the file with superslicer prevented the issue.

Using an SV06 with Mainsail OS

SoftFever, 2023-08-03:
This should be fixed in 1.6.4-beta 
https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.4-beta

piotrb, 2023-08-23:
ooh .. I gotta turn off my speed workarounds and see if I get a shutdown :) Thanks for attempting a fix here :)

piotrb, 2023-08-24:
Hmm this still happened to me when I removed switched back to more in line with the default voron profile speeds on 1.6.4-beta3 .. trying again on 1.6.6 to see if that makes a difference .. 

Also wondering what this has to do with fans .. I'm printing PLA .. shouldn't that basically be 100% fans all the time anyways?

piotrb, 2023-08-25:
No, I'm definitely still getting this with 1.6.6 ..

ErdbeermarmeladebrotmitHonig, 2023-08-30:
I think that I am facing the same problem. I'm getting these "Timer too close" errors as well.

As @piotrb, I am running a multi-mcu setup and my toolhead is connected via CAN.

I got this error today, then I changed the CAN speed. Afterwards, I printed the same gcode-file again without re-slicing. During the second print, I got the same error at the same place as during the first print. I attached my gcode file and some photos, which show where the printer stopped.

[mmu3-pulley-body-R3_ASA_4h40m.zip](https://github.com/SoftFever/OrcaSlicer/files/12480697/mmu3-pulley-body-R3_ASA_4h40m.zip)

[image]

ErdbeermarmeladebrotmitHonig, 2023-09-01:
Probably,  I fixed my issue:

https://klipper.discourse.group/t/timer-too-close-on-voron-2-4/10103/13

closed 2023-12-11 by github-actions[bot]

omarquis, 2024-07-23:
@piotrb any news? I got the same issue with an Creality Ender V3 KE and a Flashforge 5M. The issue occur more often when I'm using fuzzy skin.