OrcaSlicer/OrcaSlicer #806: Note for Carbon X1 should be at the begining

chrisribe, 2023-04-20:
This note should be at the begining of the tutorial.
https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#note

I started my print, came back continued the tutorial then saw the note. Now need to re-print.
Thanks, great work by the way ;)

domik82, 2023-04-20:
@SoftFever  is there any way to get privs to the wiki github space? I did same mistake when I did it for the first time.
I have fix but I honestly don't know how to create PR - creating branch doesn't work (I got 403). I did fork but wiki is not there. 

Changes are in attached patch (fixed also anchoring and small typos).

[Fixing_documentation_issue_#806_-_move_note_before_steps_so_that_users_won't_need_to_re-ru.patch](https://github.com/SoftFever/OrcaSlicer/files/11291026/Fixing_documentation_issue_.806_-_move_note_before_steps_so_that_users_won.t_need_to_re-ru.patch)

chrisribe, 2023-04-21:
Ah seems github wikis are not managed the same.
I messed around a bit too before finding this.
https://nimblehq.co/blog/create-github-wiki-pull-request

I will implement it on my fork and if it works will create a pull request.
Fingers crossed ;)

SoftFever, 2023-04-22:
@chrisribe Awesome!

chrisribe, 2023-04-24:
> @chrisribe Awesome!

Created the pull request hope it meets your requirements :)

chrisribe, 2023-05-10:
> @SoftFever is there any way to get privs to the wiki github space? I did same mistake when I did it for the first time. I have fix but I honestly don't know how to create PR - creating branch [...]

@domik82 would you mind making the change ? Or should I ?

SoftFever, 2023-05-14:
@chrisribe @domik82 
I just updated the WIKI with the changes you requested. 9bd6fed379310e48b899c05e451164287d852e8d
Feel free to make PRs if you have a better idea to for the change.

And thank @chrisribe again for the "github-wiki-pull-request" work! It's so much better now!

closed 2023-05-15 by chrisribe