OrcaSlicer/OrcaSlicer #812: Default process crashes on startup when default process contains a layer height not in printer's limits

mdbijl, 2023-04-21:
**Description/Reproduction**
When the layer height in the Process is not within the limits of the printer (as specified in Layer Height Limits / Min/Max) as popup message will appear saying "Too large layer height. Reset to x.xx". Additionally, when the layer height in de **default** selected Process is not within the limits, it will display the popup too (just after the splash screen appear).  However when you close the popup, it will crash the app.

**Workaround**
Edit the Process or Printer JSON manually and restart app.

**Possible cause of the crash**
It looks like the UI did completely start, maybe after closing the popup some dependencies on the UI are not there yet.

**Related problem: changing the default printer not possible**
If you have copied a printer as a User Preset, it will always use the default Process from the printer it inherits from. Adding a default Process in the user preset's JSON will only work until restarting the app. 

**Affected versions**
OrcaSlicer 1.6.1 and 1.6.2 beta have been tested by me, both show the same problem.

**Suggestion**
**Desktop (please complete the following information):**
 - MacOS Monterey 12.6

derx05, 2023-05-02:
This bug drives me crazy and is "critical" in my opionion. It can easily break the software with hobbyists not knowing how to fix it.

Vindikkator, 2023-07-23:
Yes I have this problem and don't know which profile has the issue to change... So no fix other than that?

Vindikkator, 2023-07-23:
How do I reset my profiles. Can't use even older versions of orca slicer?

Buzut, 2023-09-16:
Real pain in the butt. Especially since I have several printers and had to go and modify all the .json pref files as I had no idea which printer file it was loading…

katech35, 2023-09-22:
Same issue on OrcaSlicer 1.7.0....Unable to launch the program and not able to locate which JSON file to edit to restore functionality.   @mdbijl can you share the location of the JSON file?

rchacker, 2023-11-16:
same issue, still broken, where is the json file?  MacOs.

rchacker, 2023-11-16:
no luck with a re-install.

rchacker, 2023-11-16:
/Users/yourusername/Library/Application Support/OrcaSlicer

Deleted this folder and restarted however you will lose all your settings.

aaronmrosenthal, 2024-02-11:
My machine has a 1.4mm nozzle the machine I copied had a max of 1.2mm nozzle. Slicer is dead on start. Same issue. 

Why is 1.4mm not in the drop down list?

DBShrill, 2024-03-11:
Had same issue here with 1.91 and 2.0.0

wr45, 2024-03-13:
Same issue here with 1,91 to 2.0.0

closed 2024-06-20 by github-actions[bot]