OrcaSlicer/OrcaSlicer #814: Option for different profiles for different plates

Gizmo6, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
The ability to assign different filament/ profile to different plates would be good for big multi-material projects.

**Describe the solution you'd like**
Just as you now have the ability to name the plates it would be good to be able to have everything i needed printed in tpu on one plate. Then on a separate plate have all my abs parts. This would make it easier rather than having to save a bunch of different files.

**Describe alternatives you've considered**
Even the ability to change this via the by object option is a very good idea.

BounceG, 2023-08-20:
Fully Support this idea.

mattiasthalen, 2023-08-24:
I was just about to suggest this. It’s exactly the challenge I’m facing now.

Fabietto996, 2023-09-06:
It is precisely the function I was looking for .... we hope they do this addition

closed 2023-12-14 by github-actions[bot]