OrcaSlicer/OrcaSlicer #815: 3D connexion Spacemouse no function on Mac

pandel, 2023-04-21:
I duplicate an issue for BambuStudio, hopefully it can be fixed in OrcaSlicer, too: https://github.com/bambulab/BambuStudio/issues/1462

As stated, the 3d connexion Space Mouse isn't working, though it is absolutely useful in Superslicer for positioning support blockers, enforcers and such.

Please fix it, thanks! And keep up the amazing work you put into OrcaSlicer.

Regards,
Holger

FlyingLemming, 2023-05-09:
I am currently using my Space Mouse Enterprise with macOS 13.3.1 on a Mac Studio and, consequently, the Apple Silicon version of Orca Slicer 1.6.2. Almost everything works perfectly fine, only left/right as well as up/down movements are quite slow despite having set the speed for those two axes to maximum in the device setup.

What specific device are you using? I assume you are on the latest driver version?

pandel, 2023-05-09:
Strange... Yes, I am on the latest driver version, but because you said it is working for you, I downloaded a fresh copy of the current OrcaSlicer version just a minute ago, deleted everything regarding the slicer under ApplicationSupport and started all over again. It is working now. I did not think about doing this because the SpaceMouse worked without any problems with SuperSlicer and Fusion 360.

Many thanks!

closed 2023-05-09 by pandel

leandrolima-nyc, 2023-06-01:
I can confirm that the desired functionality works flawlessly on OrcaSlicer. Specifically, I utilized the Pro version and successfully incorporated custom commands accordingly, along with increasing the speed of the up, down, right, and left axis movements.

[image]

[image]