OrcaSlicer/OrcaSlicer #816: Unable to disable timelapse

pointon242, 2023-04-21:
The "none" option from the timelapse drop down box is missing from the V1.6.2 beta vzbot download. The only options available to select are Traditional and Smooth

Coma-patient, 2023-06-28:
+1 one have this issue aswell

Coma-patient, 2023-11-06:
@pointon242 
So I was using my ender 3. I just needed a don't make timelaps option in the slicer settings. 
Kind regard Coma

tlhintoq, 2024-02-07:
> This would be resolved in #2471

FYI:  As of the nightly build version 2 06FEB2024 - Its not fixed, at least for Bambu Labs X1Carbon

[image]

WindowsX64
Bambu X1Carbon

cc: @Noisyfox  @SoftFever

HumbleDeer, 2024-02-07:
> FYI: As of the nightly build version 2 06FEB2024 - Its not fixed, at least for Bambu Labs X1Carbon [...]

@ anyone

Would be curious to know whether this is just its own Gcode or if Orca will also recognize currently present timelapse Gcodes in the pre-print Gcode settings.

closed 2024-05-15 by github-actions[bot]

franciscomirasg, 2024-12-25:
The issue persist on 2.2.0 and the default is on when sending plate to the microSD.
[image]