OrcaSlicer/OrcaSlicer #817: [Feature Request] Add support for single instance loading

Ohmarinus, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
The problem is that when an STL is exported from Fusion360 to OrcaSlicer under windows, it always opens up a new instance of OrcaSlicer.

With PrusaSlicer the fix is to run it via a bat with the --single-instance affix in the command line. Sadly, this option does not seem to exist for OrcaSlicer. It also works for Cura. So I would really like to see this option in OrcaSlicer.

**Describe the solution you'd like**
Adding the option to add --single-instance in the command line will bring a lot of improvement to the firmware.

**Describe alternatives you've considered**
I have tried adding this line in the same way I do with PrusaSlicer and Cura, but for OrcaSlicer it doesn't appear to work.

**Additional context**
What I personally do is create a *.bat file and call that file from Fusion360, this works the best.
The file contents would look like this:
`@start "" "%~dp0\orca-slicer.exe" --single-instance %* &`

noobydp, 2023-05-10:
+1 

Superslicer and Cura have Preferences in the app to control this behaviour, this would work better for me than the command line switch.

[image]

Ohmarinus, 2023-05-12:
> +1
> 
> Superslicer and Cura have Preferences in the app to control this behaviour, this would work better for me than the command line switch.

Yes, having them as a setting in-app would also fix the issue. Although I do expect the solution that PrusaSlicer uses is super easy to implement since it already exists and Orca is a forked version. I was actually surprised to find this command-line option didn't exist in OrcaSlicer.

arraylabs, 2023-05-15:
I would some this feature as well, working on a design I end up with a pile or orcaslicer windows opened which isn't ideal. Thanks!

arraylabs, 2023-06-08:
Here is a workaround in the meantime:
```
@echo off
setlocal

:: Define the name of the process.
set "appPath=C:\Users\USERNAME\Documents\OrcaSlicer_Current\orca-slicer.exe"
set "appName=orca-slicer.exe"

:: Check if the process is running.
tasklist /FI "IMAGENAME eq %appName%" 2>NUL | find /I /N "%appName%">NUL
if not "%ERRORLEVEL%"=="0" (
    start "" "%appPath%"
)

endlocal
```

julie777, 2023-06-09:
> Yes, having them as a setting in-app would also fix the issue. Although I do expect the solution that PrusaSlicer uses is super easy to implement since it already exists and Orca is a forked [...]

The preference was already in Prusa Slicer when BBL forked it. I wonder if they deleted all the code, or just removed all the settings from preferences.

arraylabs, 2023-06-10:
From a very quick look at prusaslicer code vs orcaslicer code it looks like the preferences were removed somewhere along the way. The slicer3d stuff still appears to be intact.

jbkuma, 2023-10-20:
+1 on this, it's sort of a critical feature and I already miss using Prusa Slicer now that I've picked up the P1P!

Senkii-code, 2023-11-20:
+1, loading a new instance every time one exports a model is unnecessarily time consuming. Ideal would be an option to load to existing or new plate.

npopson, 2023-11-21:
+1, I miss this feature from PrusaSlicer.

ipd-otto, 2023-11-30:
> +1, it´s really annoying to export models from fusion...

I've got the same problem. It'd be a much better workflow with single instance.

lucas577, 2023-12-04:
I was kinda disappointed when it opened a new instance of orca, I hope this get implemented aswell. +1

Lincoln-Jones, 2024-03-04:
Still very much hoping this can get addressed. Being able to export a file from Fusion and have it open in an existing OrcaSlicer instance would be incredibly convenient.

a0s, 2024-03-05:
So, how to override this in dirty way under mac?

gbumgard, 2024-03-09:
I just had to switch to using a Windows PC to run Fusion after my MacBook died. Starting a new instance of Orca every export is annoying. Definitely works better under MacOS where I can send as many bodies as I want to the same plate.

alexeyzel, 2024-03-27:
I moved to Orca from PrusaSlicer and sending files for print from Fusion to Orca is really annoying because each time it opens a new window. +1 for this feature request.

Ohmarinus, 2024-03-27:
> I moved to Orca from PrusaSlicer and sending files for print from Fusion to Orca is really annoying because each time it opens a new window. +1 for this feature request.

I'm surprised that after a whole year not a single DEV has reacted to this question as I think it's a very valuable addition. I am still missing the function quite a lot.

For now my workaround is to directly export my entire project as a STEP file from Fusion360 and then import the entire design and pick the parts that I want to print. But being able to force Orca into a single instance is so much better.

Ohmarinus, 2024-03-27:
> It seems that the best solution here would be to open the file on a new "Plate" in the same project in the same application instance. To a first approximation, OrcaSlicer should be able to receive signals one way or another about opening files (dbus is not in MacOS - how do other programs do this?)

I'm not too experienced with these things but I work on Windows 10 and it seems that this is how it works in Windows:
https://learn.microsoft.com/en-us/uwp/schemas/appxpackage/uapmanifestschema/element-instancing

ruddy17, 2024-03-30:
This is a must-have feature! Also have been waiting for it since day 1 of transitioning to orca. It is implemented in all other slicers, including prusa, which orca was forked from. Duh.

noobydp, 2024-03-31:
This will be resolved by #4810

closed 2024-07-07 by github-actions[bot]