OrcaSlicer/OrcaSlicer #820: estimated end time

Tomek5111, 2023-04-22:
Hello 
I use the OrcaSlicer with my sidewinder x2 and klipper.  Unfortunately, the print time of the slicer is wrong and is wrong by about an hour for small prints and even several hours for large ones.  Can someone tell me what is wrong?  Is it because of the Klipper or the slicer?  The slicer controls the acceleration and speed..
[image]

uvalml, 2023-07-11:
have you solved it?

discip, 2023-09-21:
@Tomek5111 See #2003.

closed 2023-12-29 by github-actions[bot]

RyanDevineD, 2024-06-06:
Is there a way to add an offset to the estimated time, I find Orca is off by about 20% in most cases for me.