OrcaSlicer/OrcaSlicer #822: M73

Ziv831, 2023-04-22:
**Describe the bug**
Hi
I have an home made cr-10 like printer, the problem is whene the g-code is generated M73 is give me error while printing, but the printer is still printing perfectly, is there's a way to cancel the M73 code in the slicer while generating the g-code?
10x
Ziv

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
Home made like cr-10
**Desktop (please complete the following information):**
 - OS: windows 11
[image]

 - Version [e.g. 22]

zerog2k, 2023-06-14:
+1
I also have Artillery Sidewinder X1 with the newer Digant firmware (marlin 2.x), and the g-code output from orcaslicer seems to add regular M73, but this makes for annoying status popups and audible beeps every layer.
I'm struggling where to find how to disable in orcaslicer settings (I'm very new to this lineage of slicer, coming from Cura).

gerards, 2023-06-18:
Same issue with the BIQU BX, very annoying having the frequent beep and unknown command message. 

In SuperSlicer 2.4 I've disabled the M73 gcode by un-checking the "Print remaining times" option. Would be great if this can be implemented in Orca Slicer,

viruha, 2023-07-25:
Same issue with the Artillery Genius.

GBPrinting, 2023-10-14:
I am having the same problem on Biqu BX. Is there a soultion?

gerards, 2023-10-14:
@GBPrinting My work around on the BIQU BX was tweak Luke's firmware which I've been using (https://github.com/looxonline/Marlin).

The compiled file https://github.com/gerards/Marlin/blob/BIQUBX_ALMOST_STOCK/firmware_GS_Tweaked.bin is here as well as the changes to Configuration_adv.h and Configuration.h should you want to compile it yourself.

I enabled M73 and tweaked a few other bits for myself while there, installed it and have been using Orca ever since, love it!

You can modify the gcode file produced by Orca and remove all the M73 lines also but that gets old quick...

Prop3r, 2023-12-11:
Same here. Get frequent M73 echo messages, stating the remaining time. Please provide an option to turn off M73 remaining time in Gcode. 

Prusaslicer has this function, "supports remaining times" under Printer settings:
[image]

Armageddit, 2023-12-17:
This massive sound on the Sidewinder X1 of every M73 is very annoying!
pleas make a setting point to turn it off!

Prop3r, 2023-12-17:
If you're using Bigtreetech TFT firmware on your touchscreen, it's VERY easy to turn off ECHO messages:

MENU -> SETTINGS -> SCREEN -> UI -> ACK NOTIFICATIONS STYLE : OFF

ManniAT, 2023-12-17:
Unfortunately this also disables other (possibly important) notifications.

MBlackmagic, 2024-01-01:
I have written a python script and a windows droplet for this purpose. You can drop your gcode file onto the M73exDroplett and it removes all M73.
https://github.com/MBlackmagic/M73ex/tree/main

MBlackmagic, 2024-01-12:
[image]

Orca Slicer V1.9

robinson01, 2024-12-11:
The issue is still present in 2.2.0

closed 2025-02-16 by SoftFever