OrcaSlicer/OrcaSlicer #824: plate is missing

docemiel, 2023-04-22:
good morning. I'm using linux mint 21.1 vera. After configuring the application, what happens is that I don't have any table, whatever printer I use. I leave here a picture of the printscreen, I would like to have some help to resolve this issue. thanks

[Captura de ecrã de 2023-04-21 00-16-23](https://user-images.githubusercontent.com/105605043/233778918-f7f62548-acf4-4e42-9908-920b7e78bf57.png)

docemiel, 2023-04-22:
Good afternoon, After reading in another post that it could be a graphics problem, I decided to change the graphics drivers instead of using the nvidia ones, I opted for the open-source ones... and now it works! But it's still a bug or software conflict
[image]
[image]

SoftFever, 2023-04-28:
Thanks @docemiel 
However, I'm afraid that GPU driver compatibility on Linux is beyond the scope of this slicer.
Remember Linus's middle finger towards Nivida? @@

closed 2023-04-28 by SoftFever