OrcaSlicer/OrcaSlicer #827: Seam Painter Line Tool

jberry, 2023-04-22:
**Is your feature request related to a problem? Please describe.**
When trying to draw seams, for me it's almost impossible to hand-draw a straight line with the seam painter tool

**Describe the solution you'd like**
It would be great if there was a Line tool for the seam painter. i.e. click your start point, and a line forms from there to your mouse pointer where you go and click the end point. This would result in a perfectly aligned seam, or even the ability to create a perfectly in-a-line angled seam too.

**Describe alternatives you've considered**
There really isnt' alternatives that I can think of. 

**Additional context**
Open almost any drawing/paint tool, and they all contain a Line tool as I'm describing (just for usage/context of the tool itself)

PrimeGoat, 2023-12-25:
I have a solution for you that you can implement yourself.

1. Look into some automation apps that allow you yo reproduce keystokes and mouse actions, etc.  Find one where you can define the actions using scripting or via editable text
2. Go into the slicer and into the seam drawing mode and take a screenshot
3. Get the coordinates of your intended line's start and end, then use this information to create an automated task for click-dragging (drawing) this line in the slicer. You may have to look into finding formulas/algorithms for calculating pixels along a line. This info DEFINITELY exists.

This will allow you to draw the lines when needed, as I don't think anyone will be implementing this any time soon.

Eldenroot, 2023-12-26:
It is already implemented 👍

PrimeGoat, 2023-12-26:
> It is already implemented 👍

Ahh so I was right being that it's already implemented, then noone will be implementing it again any time soon 😅

closed 2024-04-02 by github-actions[bot]