OrcaSlicer/OrcaSlicer #830: Square tree supports?

vibinzthegreat, 2023-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

KD7LRJ, 2023-04-28:
I'm getting this also. I've had it happen on two or three projects now and I think I've found a workaround, but I'm hoping you can find and fix the issue with this info.

[image]

The top part (1.3mf) was sent to me to print. I believe it was exported from Solid Works as a .3mf and I imported it into a new Orca Slicer project.

The bottom part (2.stl) was created by exporting the first part (1.3mf) as a .STL file from Ocra Slicer and then imported into the project. That is the only difference between the files. I did not change any setting on either object individually, they are both using the project's settings.

The attached file is a renamed .3mf file (not an actual .zip file).

[Square Tree Supports.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11350424/Square.Tree.Supports.3mf.zip)

bonjorno7, 2023-08-17:
I've had this problem too.
Increasing "Tree support branch distance" seems to help.

closed 2023-11-24 by github-actions[bot]

Dustmuffins, 2023-11-30:
I'm not sure if it's possible to reopen this or not, but I'm having issues with this randomly popping up. I'm having a difficult time reproducing it though. If I have any more info I'll add it here.

[image]